gnea/grbl-LPC #1: Compile errors

xpix, 2017-03-25:
Hello,

i love this idea and want test the Homing Feature on a smoothieboard. But first of all i try to compile the grbl-lpc on my cloud9 workspace, but:

```
xpix:~/workspace

 (master) $ make
/bin/sh -c "mkdir -p \"build\""
/bin/sh -c "mkdir -p \"build/src\""
arm-none-eabi-g++ -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -fno-rtti  -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -Wno-unused-parameter  -I CMSIS_5/CMSIS/Core/Include -I CMSIS_5/CMSIS/Driver/Include -I CMSIS_5/Device/ARM/ARMCM3/Include -I grbl -I grbl-lpc -I lpc17xx -I VCOM_lib  -o build/src/coolant_control.o grbl/coolant_control.c
In file included from grbl/grbl.h:42:0,
                 from grbl/coolant_control.c:21:
grbl-lpc/debug.h:24:18: fatal error: atomic: No such file or directory
 #include <atomic>
                  ^
compilation terminated.
make: *** [build/src/coolant_control.o] Error 1
xpix:~/workspace (master) $ amke clean
bash: amke: command not found
xpix:~/workspace (master) $ make clean
/bin/sh -c "rm -rf \"build\""
xpix:~/workspace (master) $ make
/bin/sh -c "mkdir -p \"build\""
/bin/sh -c "mkdir -p \"build/src\""
arm-none-eabi-g++ -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -fno-rtti  -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -Wno-unused-parameter  -I CMSIS_5/CMSIS/Core/Include -I CMSIS_5/CMSIS/Driver/Include -I CMSIS_5/Device/ARM/ARMCM3/Include -I grbl -I grbl-lpc -I lpc17xx -I VCOM_lib  -o build/src/coolant_control.o grbl/coolant_control.c
In file included from grbl/grbl.h:42:0,
                 from grbl/coolant_control.c:21:
grbl-lpc/debug.h:24:18: fatal error: atomic: No such file or directory
 #include <atomic>
                  ^
compilation terminated.
make: *** [build/src/coolant_control.o] Error 1
xpix:~/workspace (master) $ make
/bin/sh -c "mkdir -p \"build\""
/bin/sh -c "mkdir -p \"build/src\""
arm-none-eabi-g++ -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -fno-rtti  -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -Wno-unused-parameter  -I CMSIS_5/CMSIS/Core/Include -I CMSIS_5/CMSIS/Driver/Include -I CMSIS_5/Device/ARM/ARMCM3/Include -I grbl -I grbl-lpc -I lpc17xx -I VCOM_lib  -o build/src/coolant_control.o grbl/coolant_control.c
In file included from grbl/grbl.h:44:0,
                 from grbl/coolant_control.c:21:
grbl-lpc/delay.h:42:71: warning: multi-character character constant [-Wmultichar]
     delay_loop(get_time(), uint32_t(uint64_t(SystemCoreClock) * us / 1'000'000));
```

I removed this atomic include, but run in the next error. I think i'm wrong :)

tbfleming, 2017-03-25:
<atomic> comes with newer gcc versions

xpix, 2017-03-25:
Ok, compile local in my windows system with arm toolchain works very well. One little thing:
`grbl-lpc/current_control.cpp:67:47: error: 'round' is not a member of 'std'
         uint8_t(std::min(255.0, std::max(0.0, std::round(amps * CURRENT_FACTOR)))),
                                               ^`

I renamed this to => 'round(..)' and it works ... nice.

closed 2017-03-25 by xpix

mayhem2408, 2017-05-18:
@xpix I want to compile GRBL-LPC for CoreXY. What tools did you install to get it working. I make make and gcc-arm-non-eabi installed but when I make, I get a compile error 

arm-none-eabi-g++ -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -fno-rtti -std=gnu++14  -DCORE_M3 -DLPC175x_6x -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m3 -MMD -MP -mthumb -Os -Wall -Wextra  -Wno-unused-parameter  -I CMSIS_5/CMSIS/Core/Include -I CMSIS_5/CMSIS/Driver/Include -I CMSIS_5/Device/ARM/ARMCM3/Include -I grbl -I grbl-lpc -I lpc17xx -I VCOM_lib  -o build/src/coolant_control.o grbl/coolant_control.c
In file included from grbl-lpc/avr/interrupt.h:3:0,
                 from grbl/grbl.h:31,
                 from grbl/coolant_control.c:21:
**lpc17xx/LPC17xx.h:99:101: fatal error: core_cm3.h: No such file or directory**
compilation terminated.
make: *** [build/src/coolant_control.o] Error 1

cprezzi, 2017-05-18:
Before you can make, you need to run 'git submodule init' and 'git submodule update' (as written in README.md)

hodgeac, 2017-06-11:
I had the same error regarding 'round'.
grbl-lpc/current_control.cpp:67:47: error: 'round' is not a member of 'std'
No relation to submodule updates in this case as neither of the submodules for GRBL-LPC include GCC and this was a known GCC bug. Need to make sure you're using the latest version of GCC (arm-none-eabi-*).
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
Download and install the toolchain. If windows 10, manually add path to the toolchain to your PATH. If windows 7, you can probably just select "add to path" option in the installer. Today, the most current version installs to "C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q1-update", which is what you add to PATH. This is the proper way of fixing this error. Hope this helps someone! I wish the build documentation for GRBL was better. It's terrible for TinyG and G2 as well.