gnea/grbl-LPC #3: Homing not tested

xpix, 2017-04-08:
Hi,

This means the input pins for switches are disabled or just not tested?

Br Frank

tbfleming, 2017-04-08:
Not tested.

xpix, 2017-04-08:
Ok, here my change, works for me :)

````
--- a/grbl/config.h
+++ b/grbl/config.h
@@ -660,9 +660,9 @@

 // Define homing/hard limit switch input pins and limit interrupt vectors.
 // NOTE: All limit bit pins must be on the same port, but not on a port with other input pins (CONTROL).
-#define LIMIT_DDR         LPC_GPIO0->FIODIR
-#define LIMIT_PIN         LPC_GPIO0->FIOPIN
-#define LIMIT_PORT        LPC_GPIO0->FIOPIN
+#define LIMIT_DDR         LPC_GPIO1->FIODIR
+#define LIMIT_PIN         LPC_GPIO1->FIOPIN
+#define LIMIT_PORT        LPC_GPIO1->FIOPIN
````

tbfleming, 2017-04-08:
Whoops. Good catch; port was wrong.

closed 2017-04-08 by tbfleming

00alkskodi00, 2018-07-19:
Hi how can i compile the firmware, here is a manual but is not working for me https://github.com/gnea/grbl-LPC/wiki thank you

cprezzi, 2018-07-20:
@00alkskodi00 Please open a new issue instead of capturing an old closed issue!

commit 5a862a5e (2017-04-08): Fix #3: wrong port for limit switches