gnea/grbl-LPC #5: grbl-lpc on LPC1768 at 100MHz

TomKeddie, 2017-05-23:
Hi Claudio,

I'm looking around for some hardware to retrofit into a laser cutter and use laserweb/grbl-lpc.  I hope you have a moment to answer my question.

 From looking at the grbl-lpc code it looks like it runs at 120MHz.

#define XTAL        (12000000UL)        /* Oscillator 
frequency               */

This would imply an LPC1769 is needed and the LPC1768 is not fast enough (or code changes are needed for 100MHz).

I see mention on reddit that it works on some 1768 boards, should I adjust the above for 100MHz perhaps?

https://www.reddit.com/r/hobbycnc/comments/6b9p48/grbllpc/

Many thanks,
Tom

cprezzi, 2017-05-23:
I guess you are right, but I'm not the LPC specialist here.
@tbfleming Can you step in with your deeper knowlede please.

tbfleming, 2017-05-23:
I'll try to explain later when I have time. The quick answer is: no, it's complicated enough that you need to use one of Arm's wizards to edit the headers to change clock frequency.

cprezzi, 2017-05-23:
At least I can say that grbl-LPC works on the MKS SBase board (LPC1768) without changes ;)

tbfleming, 2017-05-23:
MKS used a '68? Lame! There's not much price difference between the two chips.

tbfleming, 2017-05-23:
@cprezzi if you have a scope handy, I'd be curious to hear if the step rate is coming out correctly. I think I know what MKS might have done; if so the normal build would throw off the timers instead of overclocking the chip.

TomKeddie, 2017-05-23:
As far as I know this is the mks sbase schematic, can't recall where I got it.

[MKS SBASE SCH_V12.pdf](https://github.com/gnea/grbl-LPC/files/1022723/MKS.SBASE.SCH_V12.pdf)

cprezzi, 2017-05-23:
Thanks @TomKeddie! In this schematics, it is a 1769, but on my board (from aliexpress) it is a 1768 (see https://de.aliexpress.com/item/3D-printer-32bit-Arm-platform-Smooth-control-board-MKS-SBASE-V1-3-open-source-MCU-LPC1768/32808554815.html?spm=2114.010208.3.105.ZAPwvj&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10065_10151_10068_5010018_10136_10157_10137_10060_10138_10155_10062_10156_437_10134_10154_10056_10055_10054_10059_303_100030_100031_10099_100029_10103_100028_10102_10096_10147_10052_10053_10050_10107_10142_10051_5190018_10084_10083_10080_10082_10081_10177_10110_519_10111_10112_10113_10114_10037_10183_10182_10185_10032_10078_10079_10077_10073_10123_142,searchweb201603_9,ppcSwitch_2&btsid=eb0109f2-bf6d-4f7e-b989-e6d79a5c960e&algo_expid=00266682-b810-47f2-8c53-95e02b007510-13&algo_pvid=00266682-b810-47f2-8c53-95e02b007510).

tbfleming, 2017-05-23:
The schematic is missing a critical value: Y1's oscillator frequency. The ARM wizard needs to know this value to modify the header file to set the PLL config bits correctly. The PLL converts the input oscillator to the various clocks within the chip. The wizard figures out how to configure the PLL to do that correctly.

2 possibilities:
* The MKS board has a lower frequency crystal designed to work with the PLL config that Smoothieware and GRBL-LPC use. If they did this, then grbl-lpc's current config won't overclock the chip, but the counters will run slow.
* The MKS board uses the same crystal as a Smoothieboard. In this case grbl-lpc's current config overclocks the '68 to run at the '69's speed.

cprezzi, 2017-05-24:
@tbfleming Y1 on my board is 12.000 MHZ ;)

tbfleming, 2017-05-24:
@cprezzi that means we're currently overclocking your chip.

cprezzi, 2017-05-24:
@tbfleming I didn't realize any problems yet, but I will check if the MCU gets hot.

TomKeddie, 2017-05-24:
@cprezzi I'm not sure it's like that for small micros.  The LPC1768 and LPC1769 are likely the same chip tested and sorted according to the testing.  I think it's more likely the chip will misbehave if the external environment makes it hot.

Perhaps we should tag this issue as an enhancement and plan to generate an alternate 100MHz config that people can switch in the Makefile?