gnea/grbl-LPC #6: LPC boards

chamnit, 2017-05-23:
@cprezzi @tbfleming : I'm unfamiliar with LPC boards. Can you provide a list of compatible LPC boards supported? If I understand correctly, there are others like Panucatt's Azteeg X5, which I can easily request for. I think it'll probably be a good idea for me to have one on hand.

tbfleming, 2017-05-23:
It looks like you'll have to disable current control on the X5. Its spec sheet says: "Digital pot stepper driver current control (Panucatt drivers only)"

cprezzi, 2017-05-23:
I was able to test with Cohesion3D Remix (without i2c current chip) and MKS SBase (with i2c current chip).
Other users reported, that it basically works with original Smoothieboard, C3D mini and the Panucatt Re-ARM board (with RAMPS), but I'm not sure if pin mapping is completely correct. 

The Azteeg X5 has slightly different pin mapping than a smoothieboard, but should theoretically also work.

flatbroker, 2017-06-06:
Has anyone been able to load the GRBL-LPC firmware on a X5 mini v3?  I've tried with a 4gb formatted as fat and fat32.  At power up the 'halt' light comes on and stays lit, lights flash 1, 3, 1 and go out.  'Halt' stays lit.  It thinks the firmware is loaded because it writes FIRMWARE.CUR back to the card.  Laserweb4 can see a com port, but when I try to connect it says no recognized firmware found.  The X5 mini v3 will accept all of the Smoothie firmware versions.  What am I doing wrong?  Thanks for your help in advance.

tbfleming, 2017-06-06:
@flatbroker see if you can talk to it using a terminal (e.g. PuTTY).

cprezzi, 2017-06-06:
The Azteeg X5 mini doesn't have a i2c current chip, so you need to use the version without i2s support (from my repo https://github.com/cprezzi/grbl-LPC/releases or compile yourself), otherwise the firmware cannot boot.

flatbroker, 2017-06-06:
@tbfleming, No luck with terminal communication.
@cprezzi, I tried both versions.  Neither will boot.  I tried the non i2c version with the external vref pads jumpered and un-jumpered on the Panucatt 6128 modular stepper drivers.  I also tried loading the firmware with no modular drivers installed.  I tried each iteration with 5v supplied by both usb and the board.
I'm curious as to other's results with the X5 mini v3.
I definately want to speed up engraving speed so I ordered an MKS Sbase to try.

cprezzi, 2017-06-07:
@flatbroker I'm sorry. It's hard to help without beeing able to test. There must be something that still prevents the firmware from booting. It seems that the pin mapping is slightly different to the Smoothieboard. If you send me your Smoothieware config file, I can try to compile a special version for you.

flatbroker, 2017-06-07:
@cprezzi It is attached.  Thanks for your time and help.
[config.txt](https://github.com/gnea/grbl-LPC/files/1059324/config.txt)

flatbroker, 2017-06-10:
@cprezzi 
It is working on the mks sbase v1.3.  I can communicate via terminal and Laserweb3, but I get "No supported firmware detected.  Closing port" on Laserweb4.  Any ideas?

flatbroker, 2017-06-10:
@cprezzi 
I forgot to mention Laserweb4 reports "Machine connected" right before I get the "No supported firmware detected" message.  I'm running your 3 axis i2c chip version.  Thanks

flatbroker, 2017-06-11:
@cprezzi 
I loaded Laserweb4 on two additional Windows 10 laptops and started trying usb ports on each of them.  Low and behold the first try on the third laptop connected to grbl-lpc!  No other ports on this computer will connect.  The timing must be perfect on this port.  
If you get around to compiling a version for the X5 mini pinouts, let me know.  In the mean time I'll keep working with the MKS Sbase and try to learn how to compile a version for the X5 while I'm at it.
Thanks for your help.

closed 2017-06-12 by cprezzi