gnea/grbl-LPC #7: grbl-LPC on AZSMZ Mini Board

nncc, 2017-07-02:
Hello together,
It's my first time setting up Laserweb4. I am having trouble flashing LPC to the AZSMZ Mini board (Smoothieware works fine) and am wondering if this board has been tested before by you?

When I copy the firmware.bin to the SD Card and boot the board LED1 and LED2 are blinking and after that no LED is visible at all. The filename on the SD changes to FIRMWARE.CUR, but it is not possible to connect via SERIAL. Any ideas would be very helpful!

Thank you in advance,
Nico

tbfleming, 2017-07-02:
Ick. Yet another Chinese knockoff board with no documentation available. I think you're the first.

nncc, 2017-07-02:
That's what i was fearing. Thank you for the fast reply! 
The guys with the DICE 3D printer are using it quiet succesfully (https://goo.gl/j7R9Jk) and as it is easier to get in Germany then the original Smoothie (wicked world) i just ordered it.
I guess getting the Pinout is hopeless, right?

tbfleming, 2017-07-02:
I have no idea. Maybe you could ask the vendor for a diagram and post it?

nncc, 2017-07-02:
I'm crying right now. The 'official' documentation is a photo on facebook with awful resolution :D [https://scontent.ftxl1-1.fna.fbcdn.net/v/t31.0-8/17991356_1489851384360909_1428380286592461201_o.jpg?oh=5b04062b8ee28205d0bd25ad9b53242d&oe=59C919F6](url)

nncc, 2017-07-02:
I feel ashamed now! After uninstalling the Smoothieware USB Serial driver, Win 10 took over and I am now able to connect via Laserweb4. So i can confirm grbl-LPC is working with my China-Clone. Thank you tbfleming!

closed 2017-07-02 by nncc

cprezzi, 2017-07-02:
Config.txt of smoothieware should help for the pin definitions at least.