gnea/grbl-LPC #8: Homing does not move motors

plushvoxel, 2017-07-09:
**Firmware and Board Version**
grbl 1.1f
MKS-SBASE v1.3

**Input and Output**
```
$$

 $0=1
 $1=25
 $2=0
 $3=3
 $4=0
 $5=0
 $6=0
 $10=0
 $11=0.010
 $12=0.002
 $13=0
 $20=0
 $21=0
 $22=1
 $23=1
 $24=25.000
 $25=500.000
 $26=250
 $27=5.000
 $30=1000
 $31=0
 $32=1
 $33=1000.000
 $34=0.000
 $35=0.000
 $36=100.000
 $100=160.000
 $101=160.000
 $102=160.000
 $110=24000.000
 $111=24000.000
 $112=24000.000
 $120=1500.000
 $121=1500.000
 $122=1000.000
 $130=300.000
 $131=200.000
 $132=50.000
 $140=0.400
 $141=0.600
 $142=0.000

$H

 ALARM: 9 - Homing fail. Could not find limit switch within search distance. Defined as 1.5 * max_travel on search and 5 * pulloff on locate phases.
```

**Expected Behavior**
Motors moving in the -X and +Y direction until end stops.

**Observed Behavior**
Motors not moving at all.
Motors "locking" and making high pitch noise.

tbfleming, 2017-07-09:
Do G0 and G1 moves work?

plushvoxel, 2017-07-09:
Yes. everything else seams to work. We also hat successful laser cuts and engravings.

tbfleming, 2017-07-09:
Connect to it using a terminal (e.g. PuTTY) and send it the ```?``` command. It should send a status block back. Look for ```|Pn:```. Within that field will be ```X```, ```Y```, and ```Z``` values if it sees any switches active.

Repeat when you're holding down each switch to make sure they register properly.

plushvoxel, 2017-07-09:
With the Head **in** the homing position:
```
$5=1
  ok
?
  <Alarm|WPos:0.000,0.000,75.000|FS:0,0|Pn:PXYZ>

$5=0
  ok
?
  <Alarm|WPos:0.000,0.000,75.000|FS:0,0|Pn:P>
```

With the Head **not in** the homing position:

```
$5=1
  ok
?
  <Alarm|WPos:0.000,0.000,75.000|FS:0,0|Pn:PZ>

$5=0
  ok
?
  <Alarm|WPos:0.000,0.000,75.000|FS:0,0|Pn:PXY>
```
A switch in the Z direction is not attached.

tbfleming, 2017-07-09:
The Z switch input might be confusing it. Try tying it off to either VCC or GND (opposite of what your switches drive to).

tbfleming, 2017-07-09:
Scratch that. It looks like you need to change config.h so it won't try homing Z:

```
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
// #define HOMING_CYCLE_1
// #define HOMING_CYCLE_2
```

plushvoxel, 2017-07-09:
Ah, in the `grbl/config.h`. It would be nice if this was optionally / disabled in laser mode or configurable via serial commands.
Shall I open an issue?

tbfleming, 2017-07-09:
Some lasers use a Z axis to adjust focus based on stock height. I'm trying to avoid moving too far from grbl-328's way of doing things to make it easier to move grbl-lpc to grbl-hal when it comes out.

plushvoxel, 2017-07-09:
My `grbl/config.h` now has the lines 106 to 109 as follows:
```
// #define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
```
Works for me and our K40 laser cut.

Much Thanks!

closed 2017-07-09 by plushvoxel

X3msnake, 2017-07-10:
@Marble

You can add this info to the wiki if it is not there yet it would help out
others.

Best regards

plushvoxel, 2017-07-10:
What wiki do you mean?
The [cncpro wiki](http://cncpro.co/index.php/65-documentation/initial-configuration/firmware/grbl-lpc-1-1e)?

X3msnake, 2017-07-10:
The GRBL wiki

00alkskodi00, 2018-07-19:
Hi how can i compile the firmware? Here is a manual but is not working for me https://github.com/gnea/grbl-LPC/wiki

i need to z homing ane chenge output voltages for pwm lasere