gnea/grbl-LPC #11: Open drain outputs

jimfong1, 2018-04-21:
Is there a easy way to set the LPC pin config to open drain output for using external stepper drivers with optoisolators?

Thanks

Jim

tbfleming, 2018-04-22:
You'll have to add initialization code to do it. Be careful though; if I remember right 5V optos will fry the device without a voltage converter. If you use a voltage converter IC, then you won't need open drain.

jimfong1, 2018-04-22:
I would like to set the Z axis to open drain 
Step pin is LPC_GPIO2 pin 3 
Dir pin is LPC_GPIO0 pin 20

So the init code is something like this, I think....
LPC_PINCON->PINMODE_OD2 |= 1<<3;   //Bit P2.3 is open drain
LPC_PINCON->PINMODE_OD0 |= 1<<20;   //Bit P0.20 is open drain

I'm using both smoothieware and grbl-lpc, switch back an forth between them. Grbl works faster for laser rastering.  I have a external driver on Z and its easy to set open drain in smoothieware by appending "o" in the config.txt pin configuration. Z driver works fine with smoothieware set to open drain.   However would like to set grbl-lpc for open drain too so there would not need additional hardware changes.  Not sure if the above code snippet is correct.

tbfleming, 2018-04-22:
It looks correct.

jimfong1, 2018-04-23:
Thanks I got it working.  Had to figure out the pinmode selection to turn off the internal pullup, which is default on.

moors7, 2019-04-02:
Sorry to resurrect this, but I was wondering if one of you could share what exactly needs to be changed in order to enable these open-drain outputs, and where these changes need to be made. Currently running a smoothieboard on a DIY laser, all external drivers. I think I've damaged my Alpha driver, so using the Gamma one for X. I've swapped the pin mappings in cpu_map.h, but can't seem to find where I'd need to make changes to enable the open-drain configuration. 
Any and all assistance would be greatly appreciated. Let me know if you require additional information.

jimfong1, 2019-04-02:
https://embeddedtronicsblog.wordpress.com/2018/12/27/grbl-lpc-c3d-laserboard-firmware-builds/

At the end is the link to the modified source code for open drain.

closed 2019-11-13 by cprezzi

johnsonm, 2020-01-09:
@cprezzi 's grbl-LPC now has native open drain support, based on the open drain support in @jimfong1 's code but without all the other changes:
https://github.com/cprezzi/grbl-LPC/commit/47248cb1b82671ca311967a1fcf1adbf6978dd0b