gnea/grbl-LPC #12: Laser PWM PIN how can i find it?

00alkskodi00, 2018-07-15:
Hi, how can i find it PWM pin on the [MKS Sbase v1.3](https://ae01.alicdn.com/kf/HTB1j9yDRpXXXXXOaFXXq6xXFXXX9/MKS-SBASE-V1-3-3d-drucker-controller-32bit-Arm-plattform-Glatte-steuerkarte-open-source-MCU-LPC1768.jpg)? 

And hawe this board pin for TH or TL for laser activation on CO2 PSU?

Thank you.

cprezzi, 2018-07-16:
For MSK Sbase is suggest to use the special compiled version for the MKS board from my fork: https://github.com/cprezzi/grbl-LPC/releases

Laser PWM is on pin 2.5. Connect the mosfet negative pin to TL in on PSU.

mohoxo, 2018-11-14:
I am very grateful to cprezzi for her efforts. These days I am trying to run GRBL_LPC firmware on SMOOTHIEWARE. I don't know if my line is right. I hope cprezzi can remind me. There is also (.Env.txt) whether this is a text file. Thank you!
[image]

cprezzi, 2018-11-14:
@mohoxo Your picture is ok so far (if you use a MKS SBase board and the matching firmware from my repo). 

If you like to additionally adjust the maximum laser power with a potentiometer you can connect a pot of around 10kOhm instead of the direct connection between pin IN and 5V on the laser power supply.

cprezzi, 2018-11-14:
@mohoxo Hint: The environment file sould be named exactly ".env" (no .txt at the end).

mohoxo, 2018-11-16:
@cprezzi    Thank you very much for your reply. I will use the firmware that you provide for MKS_SBASE. Carry out our test. Now the problem is.ENV  file, which may be my understanding. Can you send this file out? Let me download it. I will be very grateful.

cprezzi, 2018-11-16:
I think it is important to explain for other readers that the .env file has nothing to do with grbl-LPC. It is a part of LaserWeb.

@mohoxo The .env file is only needed when you have to change default behaviour. You don't need it if you connect the MKS board with a USB cable. For details about the .env file, please check https://github.com/LaserWeb/lw.comm-server/wiki/Environment-File-(.env).

mohoxo, 2018-11-30:
Hello!
 Now my motherboard has been successfully connected to the computer. I connect PWM according to PIN2.5 interface. But now I measure the voltage here at 10V. But the PSU I use should be 5V input, how to adjust this. Thank you!
[image]

cprezzi, 2018-11-30:
You have to use the left (negative) pin of the screw terminal. It's a mosfet that ties the pin of the laser PSU to GND. Do not connect the + pin.

schme16, 2019-04-28:
@cprezzi I've wired mine as per this discussion, and have used the MKS SBase specific firmware, but I can't see where it's said what the laser/spindle enable pin is. I'm still  a new to grbl and it's a little foriegn to me (coming from marlin), so if I've missed a config point let me know!

cprezzi, 2019-04-29:
@schme16 We don't use laser/spindle enable, just PWM pin (which is defined on the release description. Depending on your machine you have a switch for laser enable (like on K40).

schme16, 2019-04-30:
Ahh, so for a K40 setup I'd add a physical toggle for laser enable, and
grbl will just send a 0 power pwm when not making cutting moves?

cprezzi, 2019-04-30:
The K40 usualy already has a "Laser enable" switch and yes, laserweb sends PWM 0 when laser should be off.

schme16, 2019-04-30:
@cprezzi That great! Thank you so much for clarifying, I really appreciate it!
I wasn't sure if it was OK for the K40's to have the laser always enabled even at 0 PWM, but it's obviously just fine if you guys implement it like this, I know how much care is taken to do things right, here

PS: apologies for the minor thread hi-jack!

mohoxo, 2019-04-30:
@schme16    
SMOOTHIEWARE extensions should be nice. I tried many times to refresh the grbl-lpc firmware onto the MK SBASE motherboard. With Laserweb controls, can connect. And can modify the configuration file, but engraving time machine is not working. I don't know why. If your K40 upgrade is successful. Could you please share your plan for me to learn?

myk3y, 2019-10-18:
> @cprezzi That great! Thank you so much for clarifying, I really appreciate it!
> All the more seeing how badly the Smoothieware forums treat people who've purchased this board (MKS SBase) without knowing that it was close sourced (even though I'm coming from 3d printers I didn't know the ins and outs of all boards, and just selected one in my budget...) and not well liked in the SW community.

It's not so much that you are being 'badly treated', it's that you are wanting free support from Smoothie when you should be getting your support from MKS, who you have enriched.

Of course MKS isn't giving you any support, because all they're interested in is to make money off someone else's effort.

I understand you didn't know that when you bought it, but your lack of research still doesn't compel the Smoothie folks to help you.

There's a ton of information freely available on the Smoothie wiki for you to solve pretty much any problem, if you apply yourself to learning the configuration parameters. They're very well documented.

I'm not berating you, I have 3 x genuine smoothie boards and 2x MKS Sbase V1.3s myself, but I don't require much in the way of support as I have taken the time to learn how my hardware and firmware works.

Personally, I think they're being petty and should support their software, regardless of what you run it on. 

I'm pretty sure if I built my own controller, they would.

cprezzi, 2019-10-18:
@myk3y I see it a bit different. Smoothieware is not the same as Smoothieboard! Smoothieware is open source and it's alowed to be used on Smoothie compatible boards. Shure, it would be corerct if MKS defs would contribute to Smoothieware, and would support thier users, but I don't like to punish someone that has a Smoothieware question just because he useses a compatible board from a manufactorer that doesn't contribute.

All people that have Smoothieware questions (also with non original Smoothieboards) are welcome on https://forum.makerforums.info/

closed 2019-11-13 by cprezzi