gnea/grbl-LPC #17: Setting up an MKS board - PWM/TTL

FlailAway, 2019-05-23:
Hi, I posted this [(MKS)](https://github.com/cprezzi/grbl-LPC/issues/22) a few minutes back then found this forum.

Can someone please assist me with this.

Is the TTL with pin 1.23 option locked at 0 to 5vdc PWM or what? Yeah, I am a neophyte with this. :)

I do not want to fry the Laser internal electronics.

Thanks

cprezzi, 2019-05-24:
Yes, pin 1.23 is TTL which means 0..5V.

cprezzi, 2019-05-24:
Do not forget to connect GND of the MKS to GND of the LPS.

FlailAway, 2019-05-24:
> Do not forget to connect GND of the MKS to GND of the LPS.

Thank you so much for the quick replies. I am well out of my knowledge area with this and do not want to damage the Laser so I must proceed with great caution. There is no LPS, it is all contained within the metal laser housing. The laser only has two wires going to it. 
-One is the 110vac mains.
-The other has two inner wires, one-red, one-black. They must be the PWM input the manual refers to. Other than the beam window, it is fully sealed.

I have been reading up on TTL but still not clear how it works. Is it communication protocol or will it directly-control the laser intensity?

I have ordered one of these [(Frequency gen.)](https://www.amazon.com/Generator-DROK-Adjustable-Frequency-Rectangular/dp/B07KVYQ78K) with the plan of attaching it to the laser and setting to 5KHz and see if it adjusts intensity up and down.

Is that a sesnsible start?

Thanks for the help so far.

cprezzi, 2019-05-24:
I guess the red is the PWM input and the black is GND (ground). 

PWM means pulse width modulation. The signal is switchen on (5V) for a certain pulse width and then of (0V) for the rest of the time (to reach the PMW frequency). The relation between on and off time dictates the laser power. 100% on time means 100% laser power. The typical PWM frequency for grbl-LPC is 5kHz (can be changed by $33 setting).

cprezzi, 2019-05-24:
The "frequency gen." you linked does the same as grbl-LPC and is not needed in my opinion.
A simple test you could do is connecting the red/black wires to a 5V DC power supply. This should activate the laser at 100% power. (watch for safety!)

FlailAway, 2019-05-24:
> The "frequency gen." you linked does the same as grbl-LPC and is not needed in my opinion.

OK, thanks again. I was just thinking of the generator as a safe way to test as I can apply just 5vdc to the input and thus, never exceed the 10vdc limit of the laser. Even if it went to full on.

However, I got a little more brave after your first replies and I have installed the 1.23 version on the MKS and connected the 1.23 output to an oscilloscope. I am getting a square wave relative to the Power setting in LaserWeb. I have not connected the laser yet.

When running LaserWeb with gcode the oscilloscope is showing the P-P square-wave voltage of 3.40 to 3.56, is this enough? It seems to be at the bare minimum for the laser manual settings. 

I am also getting a spike when the laser run has finished and the laser would be at idle. It is 5KHz and about P-P 2.2vdc. Is that the Tickle pulse mentionend on the manual?

Tickle Pulse Signal
Tickle Frequency 5 … 5 kHz
Pulse Length… 1.0 μs ± 0.2 μs
Pulse Rise/Fall Time… < 100 ns between +0.5 V to +3.5 VDC

If not, how can I emulate that in grbl for faster switching time?

cprezzi, 2019-05-27:
It depends on your laser, if 3.4-3.56V is enough for TTL high, but usually it works. If not, you need to use a 3.3V to 5V logic level converter.

As far as I know grbl doesn't generate tickle pulses. 5kHz is the default PWM frequency. Out of your specs your laser is capable of handling up to 20kHz PWM. 
My experience shows that faster PWM is better for cutting, but slower is better for grayscale engraving and 5kHz is about the sweet spot for combined operation (at least on the cheap K40 and diode machines).

closed 2019-11-13 by cprezzi