gnea/grbl-LPC #18: M9 not working on mks

mikolo90, 2019-06-14:
Hi.
I'm running my DIY laser with mks sboard 1.3, firmware for TTL 1.23 and solenoid connected to pin 2.6 and lightburn. I've got problem with air assist - it turns on as gcode is executed, but doesn't turn off. 
If I turn off USB and power supply for rest, solenoid closes, as i turn USB first for board and later power supply for all other stuff, solenoid stays closed, but if power supply is turned on first and USB later, solenoid is opened. 
Is it configuration problem?

mikolo90, 2019-06-14:
this is gcode generated by lightburn:

; LightBurn 0.9.02
; GRBL-LPC device profile, absolute coords
G00 G17 G40 G21 G54
G90
M4
; Cut @ 50 mm/sec, 1% power
M8
G0X53Y79
G1Y118S10F3000
X90
Y79
X53
M9
G1S0
M5
G90
; return to user-defined finish pos
G0 X923 Y740
M2

In this case machine reacts to M8, but not M9.

cprezzi, 2020-02-13:
This problem has been fixed in my fork at https://github.com/cprezzi/grbl-LPC.