gnea/grbl-LPC #19: Can this run on LPC1768? SKR boards?

eightheads, 2019-09-09:
Hi, can GRBL-LPC also run on LCP1768 boards like the Bigtreetech SKR v1.3 board?

cprezzi, 2019-09-09:
The LPC1768 is not the problem, but the SKR has messd up pin mappig that would not work. GRBL needs all step (or dir) pins on the same port, which is not the case with the SKR board.

closed 2019-09-09 by cprezzi

melkati, 2019-10-17:
> The LPC1768 is not the problem, but the SKR has messd up pin mappig that would not work. GRBL needs all step (or dir) pins on the same port, which is not the case with the SKR 

I wold also like to use GRBL with a SKR V1.3

Is this a matter of using wire jumpers for wiring it correctly  or is this an architectural issue and we can definitely say goodbye to using this firmware/board combination?

PLAYTHAT, 2020-07-03:
Hello
 Is it possible to port it to skr 1.4 Turbo ???

cprezzi, 2020-07-03:
@PLAYTHAT No, it's not.