gnea/grbl-LPC #20: pn:PA triggered on MKS

dansity, 2019-11-13:
Hello,
I have wired my 3 limit switches NC ($5=1) and in the serial I'm reading `Pn:PA` and I get an alarm triggered by it obviously.
All 3 switches are recognized and when I press them and I'm reading `Pn:PXA` when pressing X so they are working.
I guess I have to use a jumper on two pins somewhere on the MKS board to close the limit of 4th axis(?)
I'm just guessing it is the 4th axis as I can see all the feeds in the settings like $113

Can I remove this with any setting in the software or can someone point me to the right pins to jumper on the MKS 1.3 board?

cprezzi, 2019-11-13:
Please use my official fork, as stated in the README.md. It has predefined profiles for multiple controllers (like the MKS SBASE), that can be activated by a simple define in config.h.
See: https://github.com/cprezzi/grbl-LPC

dansity, 2019-11-13:
I have downloaded and flashed the mks_sbase_firmware.bin from releases (94kb file - 2017 oct 8) and have the above mentioned 4th axis defined.
Am I missing something?

dansity, 2019-11-14:
In the meantime I have found that the fourth axis limit pin is defined on p1.29 (Z max)
`define A_LIMIT_BIT       29  // reuse p1.29`
I don't really know how to compile these forks so I will just short the pins and call it a day.
Can i have any future disadvantage by not having a 4th axis physically?

cprezzi, 2019-11-14:
The jumper solution is a easy one (in case of $5=1) and you should not have any disadvantage of it.

closed 2019-11-14 by cprezzi